introlab/16SoundsUSB #2: microphone RJ45

jamesshao8, 2018-07-09:
Hello there,

I made one of this 16SoundsUSB according to your design files. I found there is actually no microphone on board, instead there are 16 RJ45 sockets.

According to the schematics, there are already on board input amplifiers, so what I need is some analog microphone units, and connect their outputs to each RJ45 port 2 and 4, right?

Do you have some suggestion about the microphone model/brand?

Thank you!

Regards,
Shao

qinyunchuan, 2018-07-15:
I think you should attach a ECM to the RJ45 connector.

jamesshao8, 2018-07-15:
It's actually RJ11, I found the design file of MIC-RJ11 board from 8SoundsUSB project. I think they can be used in this 16SoundsUSB project as well. So I am trying to make them locally.  Right now the I am working on how to download firmware to this board. What is ECM anyway? A type of microphone? I do have plan to customize the MIC board with better microphone for higher sensitivity.

doumdi, 2018-07-16:
Please have a look at the microphone design here : https://github.com/introlab/xSoundsMicrophones. We have electret design and MEMS design available.

doumdi, 2018-07-16:
Please note :  We use RJ11 connectors and cables in a custom configuration, allowing differential signals and power supply to our own microphone design. It is not compatible with any external 3.5mm adapter and microphone. We made this choice to make the system robust to noise, compact and easy to install. Everything is powered by the sound card, using USB 2.0.

jamesshao8, 2018-07-16:
Nice! Will take a look at these microphone designs.

jamesshao8, 2018-08-09:
Sorry, I was wrong about the sensitivity, mems with -38dB sensitivity should be better than the electret one with -44dB sensitivity. But which pre-amplification is better is still unknown to me. On electret design I have measured the real max gain is 20 times. Will try the mems one. I might also put those amplifiers in series and see how they work together.

vrheaume, 2018-08-09:
Hello James, 

Our MEMS microphone board design is more recent and was designed to go along with our 16SoundsUSB. The electret microphone board design dates back to when the 8SoundsUSB was designed. 

The gain values for both microphones were set in a very subjective / arbitrary manner. We had 2 concerns: we don't want too much gain to saturate the signal chain, yet we want enough gain to be able to hear properly when speaking from a few meters away. 

Overall, we like the new microphones better with the 16SoundsUSB than we liked the older microphones with the 8SoundsUSB. However, we haven't done apples-to-apples comparison on the two microphone designs. 

The MEMS microphone has better sensitivity (by 6dB) and better SNR (by 2-4dB), according to its specification (but I'm not sure we can really compare the specs since they're from different manufacturers, thus maybe different test conditions). We also experienced much, much less signal degradation coming from electromagnetic interference with the MEMS microphone (which we noticed recently when using the electret microphones a few inches away from a wifi router). The VM1000 MEMS microphone is supposedly waterproof and dustproof, features we haven't explored in-depth yet. 

It is possible to change one resistor to adjust the gain on the electret microphone board, should you find it desirable. Let me know if you need help with this.

jamesshao8, 2018-08-10:
Thank your for your explanation. 

I will try to make the mems one locally and compare them with the electret one. I noticed in the mems design R7/R6=10 is the gain, and I replaced R7 to 100k resistor, hoping to get higher gain. I have also bought some mics from knowles. One of them has -18dB sensitivity and 63dB SNR, another one has -40dB sensitivity and 70dB SNR. I am worried about saturating the ADC as well, so I am not sure choosing a mic with higher sensitivity or using higher gain in amplifier would guarantee I can hear sounds for longer distance, but I think higher SNR is always better.

vrheaume, 2018-08-10:
In case you haven't noticed, you should be aware that soldering the MEMS microphone is not trivial. The trick I use is to put solder on the PCB pads, then place the microphone with the tweezers, then use the hot air gun to reflow the solder. The MEMS microphone will self-align with the pads as the solder reflows, and then I give it a small push downward with the tweezer to make sure it is flat against the board. 

If you go intend to increase the gain, you might want to make sure the opamp's gain-bandwidth specification is respected... 

Keep us updated with the results of your experimenting!

jamesshao8, 2018-08-10:
We have a partnership with a local PCB/SMT factory. We usually ask them to do small volume prototyping. I think they are able to solder this mems microphones. But for knowles mics, I will have to solder them by myself, connecting them with some wires, so that I can put into my bread board and do the experiments. 

We would love to contribute to this interesting project. So sure we will update our result.

jamesshao8, 2018-08-25:
Hello @vrheaume,

I have tried the mems mics these days. They are receiving sound in longer distance than electret ones indeed. And my own modification of the resistor increased the distance even further. But the distance is still not enough for me. And in long distance, I can't see obvious beamforming effect as I can in short distance.

Also I am wondering why a basic op-amp is used in the mems design instead of a specialized audio amplification chip in the electret? If I some how connect the mems mic to the amplification chip in electret design, will the distance be even longer? Thanks.

vrheaume, 2018-08-25:
Hello @jamesshao8,

Regarding the microphone distance: I have no precise recommendation for this, but here are a few thoughts:

- Perhaps you could get a few dB of gain with enclosure design. 
- What is the maximum gain you experimented with? Perhaps we could try with higher gain still. Perhaps this would require another amplifier due to gain-bandwidth product limitation, perhaps it would require modifications on the microphone PCB design. 
- Are you getting clean signals when you amplify more? Does it start getting noisy? If so perhaps we could find a lower-noise opamp (would cost a little bit more; perhaps something like the [OPA1662](http://www.ti.com/lit/ds/symlink/opa1662.pdf)). 
- Perhaps we could shop around specifically for microphones that are intended by the manufacturer for "far-field use"; for example [this one](https://www.invensense.com/news-media/invensense-introduces-industrys-first-74-db-snr-analog-mems-microphone-enabling-far-field-audio-capture/). 

Regarding the choice of using an opamp instead of the TS472 amplifier: I chose to do it this way because, in general, I try to use parts which have several compatible alternatives, so that we avoid sourcing problems. The opamp package is standard, you can find drop-in replacements from several manufacturers, and models with different specs (power consumption, THD, gain-bandwidth product, etc), which isn't the case for the TS472. 

I do not tend to think you'll see much difference by using the TS472 instead of the amplifier that's already on the MEMS microphone board. That would be an interesting thing to try out; let us know if you do.

vrheaume, 2018-09-05:
Yes, an enclosure/case _could_ help a little bit. I am no acoustics expert, it seems hard to tell how much improvement you could get, but I would think on the order of 1dB perhaps. 

Regarding the electronic gain: you could go about 6-10dB more, with just a slightly better opamp (such as MCP6L92) and changing R7; going above this might require a new circuit with two stages of amplification. 

However, when you increase the gain (either mechanically by enclosure design, or electronically), the tradeoff is that you'll get saturation faster. "No free lunch!"

Perhaps an AGC (automatic gain control) function (either fully-analog, or a programmable amplifier controlled by real-time processing firmware), in addition to the amplifier, should be considered. It would involve some hardware development however. Here's an [application note on the subject of AGC](http://www.onsemi.com/pub/Collateral/AN-6603.pdf.pdf).

jamesshao8, 2018-09-05:
Hello @vrheaume ,

I just tested the 16SoundsUSB and mems mic with my own modification at R7 in a silent lab (No reflection of sound, and no noise from outside the room). We found there is actually very little noise left on the sound sources figure ( which means there isn't much background noise from the circuit). We can hear clearly human voice from a distance of 8 meters. So I think the sensitivity of mic and gain of op-amp is enough. Our main problem is with the beamforming algorithm. Also if we could extend 16SoundsUSB to a 64 channel device, it should have narrower pattern with high gain. But we will always try with high SNR mic and better op-amp. Since the modification of the main boards is too hard for us.

doumdi, 2020-08-19:
@jamesshao8 @vrheaume. There is a limitation with USB2 with the number of channels we can transmit over endpoints. 16 in is close to the maximum @ 96kHz.