introlab/16SoundsUSB #3: Please tell us what you do with 16SoundsUSB

doumdi, 2018-07-16:
We are curious to know how and where you use the 16SoundsUSB device. Please let us know!

jamesshao8, 2018-07-16:
I am originally from the radio world. Always wanted to study DOA and beamforming algorithm which is used in smart antenna. But multi-channel radio equipment are usually very expensive and there aren't many existing projects to learn from. Recently I found respeaker v2.0 can be used with odas to do 3-dimensional DOA and even beamforming. Then I found this 16SoundsUSB hardware, which I think can provide better resolution since it has more channel.

I think combined with a camera, 16SoundsUSB and odas can be used as an acoustic camera. To see where sounds come from and identify them on real-time video. With beamforming in static mode, I can even choose which sound source I want to listen to, like a telescope for sound. But if put into practical use maybe I need to scale it up to 64 channel or 128 channel and use microphone with higher sensitivity. Will experiment with existing design first and maybe simply some tuning in parameters will do.

doumdi, 2018-07-17:
@jamesshao8, thanks for sharing this with us. The acoustic camera and "static" beamformer capabilities are on the TODO list for ODAS. Please contact @FrancoisGrondin for questions concerning those topics.

U0x0U, 2020-08-19:
Hi, I'm from a research institute in Taiwan, which is committed to sound-related research . 
16SoundsUSB is awesome for microphone array, and we plan to make the circuit board first.
I would really appreciate it if you could kindly give us some advice and suggestions.

doumdi, 2020-08-19:
@U0x0U you can find all fab files here: https://github.com/introlab/16SoundsUSB/releases. I would suggest you register to XMOS to get the firmware first to make sure it fits your needs. Few modifications are needed but the design is based on the 
[xCORE-200 multichannel audio platform](https://www.xmos.com/products/audio/kits). You will need to accept the license and download the sw_usb_audio-[sw]_6.15.2 firmware and use the app_usb_aud_xk_216_mc for starting.

blursj, 2020-11-04:
@doumdi Dear sir, I am debugging 16SoundUSB board made by myself, and would you please release your few modifications which are needed based on the xCORE-200 multichannel audio platform? I appreciate for your help. My email is: blursj@gmail.com

roboav8r, 2020-12-01:
@doumdi I plan on integrating the kit onto a mobile robot platform, then using it for real-time human-robot interaction & sound event recognition.