introlab/16SoundsUSB #4: What kind of cable is supported by the 16SoundsUSB?

TooSchoolForCool, 2018-07-26:
Hi there,

Our lab bought a 16SoundsUSB from your lab a few months ago. 

But now, we are trying to replace the original cable with a shorter one since the original cable is too long. 

We bought some RJ11 cables from Amazon, and [here](https://www.amazon.com/Telephone-Plugs-Conductor-inches-Silver/dp/B01LKLRGWW/ref=sr_1_2?ie=UTF8&qid=1531095701&sr=8-2&keywords=RJ11+1+ft&th=1) is the link. However, these Rj11 cables do not work at all...

I was wondering what kind of cable is support by the 16SoundsUSB, and where can I find some shorter alternative cables.

I also took a picture of our RJ11 cable and the original cable. The left one is the RJ11 bought from Amazon, and the right one is the original cable.

[image]

vrheaume, 2018-08-02:
Hello, 

Two things: 

1. The connector on the 16SoundsUSB card can physically connect both to 4-pin and 6-pin "RJ" connectors. If using 6-pin connector and cable, only the 4 center pins are actually used to carry signals. [Go to this website for additional information on RJ11 and RJ12](http://www.differencebetween.net/technology/difference-between-rj11-and-rj12/)

2. The problem you are experiencing with the Amazon-bought cable is probably that it is a "straight" cable instead of a "crossed" cable. In a straight cable, pin 1 goes to pin 1, 2 goes to 2, etc; in a crossed cable, pin 1 goes to pin 4, pin 2 goes to pin 3, etc. In our case, +5V is on pin 4 on the microphone whereas it is pin 1 on the sound card itself, so you need a crossed cable. [This is an example of a crossed cable you can buy](https://www.amazon.com/Kenable-RJ12-Cross-Cable-BLACK/dp/B0051FKFXQ/ref=sr_1_3?s=electronics&ie=UTF8&qid=1533238325&sr=1-3&keywords=rj11+crossover+cable&dpID=41LKJ6rkqxL&preST=_SY300_QL70_&dpSrc=srch). The reason for deciding to use crossed cables instead of straight is lost in time. 

You should be able to confirm that the cables you have in hand are straight (Amazon) and crossed (ours) either visually (looking at the wire colors at both ends of the cable) or with a multimeter. You could buy new cables, or buy your own [connectors](https://www.amazon.com/Monoprice-Round-Stranded-50-Piece-107315/dp/B005E2Y7JE/ref=sr_1_3?s=electronics&ie=UTF8&qid=1533239027&sr=1-3&keywords=rj11+connector&dpID=31Y18rf0aXL&preST=_SX300_QL70_&dpSrc=srch) and [crimp tool](https://www.amazon.com/TRENDnet-RJ-45-RJ-12-RJ-11-TC-CT68/dp/B0000AZK4G/ref=pd_bxgy_147_img_2?_encoding=UTF8&pd_rd_i=B0000AZK4G&pd_rd_r=GN5WMD3JA285CFQ5EPE2&pd_rd_w=jk3NY&pd_rd_wg=tv2MX&psc=1&refRID=GN5WMD3JA285CFQ5EPE2), to recycle the cables you have in hand and allow you to make custom-length cables in the future. 

I hope this helps you. Let me know if it does not. I will look over our documentation and add a note regarding the type of cable if it isn't clear already. I'm sorry you had this problem. 

Vincent R

TooSchoolForCool, 2018-08-10:
Thank you so much, Vincent!!!

I double checked both sides of the wire, I then found out their difference. And yes, the original cable is "crossed", whereas the one I bought from Amazon is not.

Again, thank you so much :).

vrheaume, 2018-08-15:
I'm glad I could help you figure this out. Let us know if you have more questions! 

(I'm closing this issue since it is resolved)

closed 2018-08-15 by vrheaume