introlab/16SoundsUSB #5: I can not compile

Yarulik, 2019-01-16:
Hello dear!

Thank you for such an interesting project, I was pleased to collect it!)

I compiled 16SoundUSB, but I cannot compile the project by changing the files as described.
An unmodified example app_usb_aud_xk_216_mc is compiled.

Additional targets can be installed from the xTIMEcomposer Studio
using the "Install New Hardware ..." option.
xmake [1]: *** [.build_2i16o16xxxxx_tdm8 / src / extensions / gpio_access.c.pca.xml.decouple] Error 1
xmake: *** [analyze] Error 2

Can anyone share a compiled project or a binary that I could check. I am an enthusiast and learn programming.

Thank you in advance!

yarulik1986@gmail.com

doumdi, 2019-01-18:
Hello @Yarulik , 

Have you tried the procedure [here:](https://github.com/introlab/16SoundsUSB/tree/master/Firmware)
With the firmware available [here :](https://github.com/introlab/16SoundsUSB/releases)

If you try compiling from XMOS source, try removing the gpio part, it is not needed.

doumdi, 2019-01-18:
In our firmware everything in gpio_access.c is commented.

doumdi, 2019-01-18:
**1) Comment line in audiohw.xc**
//#include "gpio_access.h" 

**2) Add those definitions **
```
//Modified
on tile[0] : out port p_gpio_reset = XS1_PORT_4E;
on tile[0] : out port p_gpio_leds = PORT_LEDS;

//Define the address of the codecs
#define CS5368_I2C_ADDR1      (0x4D)
#define CS5368_I2C_ADDR2      (0x4C)

//Also define write function  to both codecs
#define ADC_REGWRITE1(reg, val) {data[0] = val; i2c_shared_master_write_reg(r_i2c, CS5368_I2C_ADDR1, reg, data, 1);}

#define ADC_REGWRITE2(reg, val) {data[0] = val; i2c_shared_master_write_reg(r_i2c, CS5368_I2C_ADDR2, reg, data, 1);}

//make sure you reset both codecs with this call
p_gpio_reset <: 0x00;

//And remove reset with this
p_gpio_reset <: 0x0F;

//Also make sure you duplicate all write to both CS5368 and you will be good to go!

```

I hope this helps.

Yarulik, 2019-01-18:
Thanks for the help!

Yarulik, 2019-01-19:
I apologize in advance for machine translation!

Today I wanted to download a binary file that you gave me, but there were problems. I confused several components R77, R79, R98 during installation, put the capacitors. Because of this, the supervisor issued on TP2 2.46V. After installing the correct components, the voltage became 3.3V
 As far as I understand the RST should be LOW. And when resetting supervisors, issue a single HIGHT pulse for 140 ms and again LOW.
I have on RST (TP_2) HIGHT, while holding the button (SW1) LOW .....
The programmer works, the IDE shows it s / n, the diodes are shown.

Probably somewhere I was wrong again! I can not understand where. I would not want to burn the microcontroller, it was very difficult for me to get it.

voltage:
VDDIOR: 3.35V
VDD: 1.0V
RST_N: 3.3V
PLL_AVDD: 1.0V
OTP_VCC: 3.35V
PREREG: 5.65V
Crystal oscillator VCC: 3.3V
Я начал более подробное изучение документации и установленных сдвоенных диодов D11 "звонится только один диод", что странно. Я его поздно обнаружил и не успел поменять. Может ли это быть причиной HIGHT на RST?

Подскажите пожалуйста, что я ещё упустил????

closed 2019-01-19 by Yarulik