introlab/16SoundsUSB #6: xrun: Cannot load image, XCore 0 is not enabled

Yarulik, 2019-01-20:
Hello!

I apologize in advance for machine translation!

Today I wanted to download a binary file that you gave me, but there were problems. I confused several components R77, R79, R98 during installation, put the capacitors. Because of this, the supervisor issued on TP2 2.46V. After installing the correct components, the voltage became 3.3V
As far as I understand the RST should be LOW. And when resetting supervisors, issue a single HIGHT pulse for 140 ms and again LOW.
I have on RST (TP_2) HIGHT, while holding the button (SW1) LOW .....
The programmer works, the IDE shows it s / n, the diodes are shown.

Probably somewhere I was wrong again! I can not understand where. I would not want to burn the microcontroller, it was very difficult for me to get it.

voltage:
VDDIOR: 3.35V
VDD: 1.0V
RST_N: 3.3V
PLL_AVDD: 1.0V
OTP_VCC: 3.35V
PREREG: 5.65V
Crystal oscillator VCC: 3.3V
Я начал более подробное изучение документации и установленных сдвоенных диодов D11 "звонится только один диод", что странно. Я его поздно обнаружил и не успел поменять. Может ли это быть причиной HIGHT на RST?

Подскажите пожалуйста, что я ещё упустил????

Yarulik, 2019-01-20:
hink/16SoundsUSB/Firmware/16SoundsUSB_Firmware_REV_1_0.xe
xrun: Cannot load image, XCore 0 is not enabled

think@think-ThinkPad-T470p:~/XMOS/xTIMEcomposer/Community_14.3.3$ xflash --list-devices

Available XMOS Devices
----------------------

  ID	Name			Adapter ID	Devices
  --	----			----------	-------
  0 	XMOS XTAG-3         	hNwmAqzA	O[0]

doumdi, 2019-01-22:
Maybe this will help : https://www.xmos.com/developer/support/knowledgebase/why-do-i-get-cannot-load-image-xcore-0-not-enabled

closed 2020-08-19 by doumdi