introlab/16SoundsUSB #7: Getting Started, Flashing Firmware on Ubuntu 18 LTS

seanernestmurray, 2019-05-30:
Hello!

Our laboratory recently bought the 16soundsUSB. I'm running into trouble getting the board up and running in the ODAS Studio environment. This may be pretty basic, so please excuse the inexperience. 

I've followed the installation (from source) and configuration procedures for ODAS and have added the TCP sockets required for local communication with ODAS Studio (in the sounds16USB.cfg file, zipped below). I've installed ODAS Studio and it starts up without any issue, and added the following paths in ODAS control:
/home/sean/odas/bin/odaslive
/home/sean/odas/config/odaslive/sounds16USB.cfg

At this point I'm not sure of the next step to take in order to get ODAS Studio running with the board.

In my .cfg file, I've specified a .raw input file named "raw_16mics.raw". Where is the audio data written to this file coming from? Do I need to change the firmware for the board (so that it writes to that .raw file) and then flash it? If so, where do changes need to be made, and how do I flash it? There are instructions given for flashing firmware using OSX, but none for the operating system I'm currently using, Ubuntu 18.04.2 LTS.

PS. I also tried using a sample audio .raw file as input in my .cfg file also, however nothing happens when I try to Launch ODAS in ODAS Studio. And if I try and launch ODAS in the terminal using ./odaslive, it returns that "./odaslive is a directory".

[sounds16USB.zip](https://github.com/introlab/16SoundsUSB/files/3236747/sounds16USB.zip)

seanernestmurray, 2019-06-04:
I believe I figured out most of the issues listed above. 

For real-time data, a soundcard interface may be specified in the 'raw' section of the .cfg file. This is outlined below - no firmware needs to be flashed for this.
```
raw:
{
    fS = 44100;
    hopSize = 512;
    nBits = 32;
    nChannels = 16;

    # Input with raw signal from microphones
    interface: {
        type = "soundcard";
        card = 2;
        device = 0;
    }
}
```
One small point to note is that in order to run ALSA with the board, the PC must not be registering the input as an active audio source - i.e. audio input from the board should be disabled in general settings before trying to access the audio data in ODAS. (This may be an obvious troubleshoot, but a speedbump nevertheless.)

doumdi, 2019-06-05:
@seanernestmurray. Happy you got the system working. You are right, there is no need to flash the sound card firmware. The procedure you found is for flashing the firmware the first time, when people build the PCB themselves. The best way to get started is with sample configurations. Here is one with 16 inputs : https://github.com/introlab/odas/blob/master/config/odaslive/azimut_oma.cfg

For ODAS questions, It would be better to fill issues here : https://github.com/introlab/odas/issues or here : https://github.com/introlab/odas_web

Depending on your Linux configuration and if 16SoundsUSB is your only sound card in the system, it is possible that the sound card was already opened by another process. Thanks for sharing this problem with us.

Regards,

Dominic

seanernestmurray, 2019-06-14:
Thanks Dominic!

closed 2020-08-19 by doumdi