introlab/16SoundsUSB #8: Supported Operating Systems

Strigi7ormes, 2019-08-08:
Does 16SoundsUSB work fine on Ubuntu or previous versions of windows i.e. Windows 8, Windows 7???

Strigi7ormes, 2019-10-05:
Thanks @Junye97 . Can you please share the drivers??

Junye97, 2019-10-05:
I believe [this](https://www.xmos.com/download/USB-Audio-Class-2.0-Evaluation-Driver-for-Windows(4.13.0).zip) is the link to the file. However, you need to register a XMOS account in order to download that. 

As I have mentioned, although I have the driver installed on Windows 10, I was not able to get the sound card to work with Audacity... You can try if you wish but I would suggest you using it on Ubuntu to save you a lot of troubles.

Strigi7ormes, 2019-10-05:
Thanks @Junye97 .

closed 2019-10-05 by Strigi7ormes