introlab/16SoundsUSB #9: What is the adc pin output when no input channel is  connected for both TDM/I2S?

Utsav196, 2019-11-06:


doumdi, 2020-08-19:
@Utsav196, Not sure I understand your question. @vrheaume any idea?

SCWhite, 2020-08-20:
HI, this is a great project.
I'd like to know if this 16SoundsUSB board support I2S microphones 
(e.g. SPH0645 ) it's a I2S MEMS Microphone.

If so how can I do this ?

vrheaume, 2020-08-20:
Hello @SCWhite, 

No, 8SoundsUSB/16SoundsUSB are not adequate for "digital" microphones. Those weren't very popular back when we first started, so they weren't given much consideration; instead we went for "analog" microphones and ADCs (which hasn't been a bad choice! it works for our ManyEars sound localization project). Since then we have noticed the potential cost advantage of "digital" microphone (no expensive ADC! no signal conditioning!), but also the risk associated with it (high-frequency digital signals on a cable that we want to be inexpensive = potential EMI problems?). It is certainly something that we want to look into further, but haven't done it yet. 

To go forward, I would tend to look into application notes and reference designs by XMOS. Such as [this one](https://www.xmos.com/download/xCORE-Microphone-Array-hardware-manual(2V0).pdf) which supports PDM microphones. Perhaps they have a similar development board/schematics/firmware for I2S microphones. 

Let me know if you start something along those lines, and how we can help. What are the advantages of I2S microphones in your application?