introlab/16SoundsUSB #11: [SUGGESTION] Multichannel Audio DSP Field Mixer Recorder projectcollaboration

forart, 2020-06-05:
Hi there, as a no profit foundation devoted to spread free/open technologies to help indipendent/emerging artists’ growth, we always try to stimulate collaborations between cool projects like this:

https://hackaday.io/project/7041-multichannel-audio-dsp-field-mixer-recorder

Last but not least, we do also have our own project - called RecRack - that aims to create an hardware to be (transparently) interposed between performers and stage/PA technicians in order to capture the purest possible audio of the show.
[ a “Columbus egg” idea that basically tries to integrate a recording function inside a DI / splitter unit ]

Hope that inspires !

Marco Ravich

doumdi, 2020-08-19:
@forart, @vrheaume . Thanks for the info.

forart, 2020-09-02:
Glad to be helpful.

Here's the project public chat link where discuss about possible collaborations:
https://hackaday.io/messages/room/270407