introlab/16SoundsUSB #12: About uploading the firmware

SCWhite, 2020-11-03:
HI, this project is really awesome.

I have a few question about uploading the firmware.
due to not familiar with XMOS chips, so bare with me if i ask a silly question.

1. Do I need to get a xTAG debug adapter to upload the firmware?
or I can get it done by any JTAG-USB cable?

2. Seems that I can't use the "USB to Host" to upload the firmware, right ?

3. I'm getting error message by using windows:
"C:/Program Files (x86)/XMOS/xTIMEcomposer/Community_14.2.3/bin/xflash.exe: error while loading shared libraries: ?: cannot open shared object file: No such file or directory"
and it might cause by the JAVA runtime environment error according to this post: https://www.xcore.com/viewtopic.php?t=6938 
is there a way to use xflash alone? that will be great.

Thanks about making this cool project, looking forward to give it a go.

doumdi, 2020-11-07:
1. Yes you need the XTAG adapter.
2. Firmware upload via USB is not supported (but this can be done with a little bit of work).
3. You might ask this question to XMOS directly. I use a Mac or Linux.

SCWhite, 2020-11-08:
@doumdi 
Thanks for the reply, XTAG adapter is on it's way

closed 2020-11-08 by SCWhite