introlab/16SoundsUSB #14: Clarification on firmware patching/flashing

roboav8r, 2020-12-02:
Hi all,
I just got my hardware kit, but unfortunately I am not sure how to flash the firmware properly. I understand the concept, but am not sure of which **specific** commands to run to apply the patch & flash the firmware onto the kit.

My OS:
- Ubuntu 20.04

From XMOS, I have:
- [sw_usb_audio-[sw]_6.15.2](https://www.xmos.ai/download/sw_usb_audio-%5Bsw%5D(6.15.2).zip)
- [xTIMEcomposer 14.4.1](https://www.xmos.ai/file/xtimecomposer-community_14_4_1-linux64-installer?version=latest)

From this repository, I have:
- [16SoundsUSB_Firmware_REV_1_0.xe](https://github.com/introlab/16SoundsUSB/releases)
- The firmware [patch](https://github.com/introlab/16SoundsUSB/tree/master/Firmware/app_usb_aud_xk_216_mc)

And so far, I have done the following:
- Set up udevrules for the device, as in section 2 [here](https://www.xmos.ai/download/Enable-USB-drivers-on-Linux(X2612A).pdf). When I connect it to a USB2.0 port, `lsusb` shows `ID 20b1:0008 XMOS Ltd 16SoundsUSB Audio 2.0`.
- Sourced the shell script `source .../xTIMEcomposer/Community_14.4.1/SetEnv.sh
- `.../xTIMEcomposer/Community_14.4.1/bin/xflash .../16SoundsUSB_Firmware_REV_1_0.xe`

The last command results in the following error:
`Error: F03136 No devices attached - Cannot run the enquirer until a device is found`

Beyond this, I'm not sure which commands to run or files to edit. Could anyone provide some more specific information about how to apply the patch and flash it to the kit?

Thanks!

doumdi, 2020-12-07:
Hello John,

Could you first tell me why you need to flash the firmware since it was already done when you purchased the kit?

doumdi, 2020-12-07:
The device requires no driver and Linux will recognize it as an USB sound card. You can use Audacity or any other audio tool compatible with ALSA for record/playback to the device.

roboav8r, 2020-12-08:
Thanks @doumdi! I didn't realize the kit had the firmware flashed already.

closed 2020-12-08 by roboav8r