introlab/16SoundsUSB #16: About using 16SoundsUSB

SCWhite, 2020-12-29:
I finally get the firmware uploaded (at least I think I do)

by following the instruction from "16SoundsUSB/Firmware/README.md"
I flashed "16SoundsUSB_Firmware_REV_1_0.xe " on my board

here comes the question, my Linux do not recognize 16SoundsUSB as a sound card
and I can't see the device by using "lsusb" or "audacity"

how do I know if this is done correctly? It should be recognize as a sound card, right?

also, after plugin to the computer, 4 leds keep lighting up
this does not match the description from #13  

>by @blursj
> when the usb power supply is plugged in, LEDs 1~3 blink 4 times and then go out.

any advice or suggestions?  

Thank you very much.

I'm using:
OS: Ubuntu 16.04
Version: xTIMEcomposer Community_14.2.3
Board: 16SoundsUSB Rev1.0

and the board looks like this:

[image]

doumdi, 2020-12-30:
@SCWhite Try to upload the code from XTimeComposer using the XTAG adapter and verify if you get error messages when USB starts to connect.

doumdi, 2020-12-30:
@SCWhite, The first thing I would verify using a scope is the I2C communication to CS2100-CP-CZZ (Clock generator), CS4384 (DAC), CS5367 (ADC x2). Check for NACKs, transmission errors, etc. Make sure the two CS5367s are on different I2C addresses (PIN 37, 38). The firmware will fail (assert) if I2C communication does not work and restart. The codecs are configured after USB is initialized by Linux usb_audio driver and sets the sampling rate. So you need to plug USB to the computer for testing.

SCWhite, 2020-12-31:
@doumdi  Thank you very much.
 I'll test them immediately.

SCWhite, 2020-12-31:
OK, I got it working on my device. It can be recognized as a usb device.
Thanks for pointing out to check I2C, you are the best.
(but the led is still strange though)

Another question, is this change from issue #13 necessary for the project ? 
file: "audiohw.xc"

>from  @blursj
>1.
>void AudioHwInit(chanend ?c_codec)
>{
>#if !(defined(SPDIF_RX) || defined(ADAT_RX)) && defined(USE_FRACTIONAL_N)
>    /* Output a fixed sync clock to the pll */
> // configure_clock_rate(clk_pll_sync,100,100/(PLL_SYNC_FREQ/1000000));
>    configure_clock_rate(clk_pll_sync, 100, 100);//100000000/PLL_SYNC_FREQ);
>    configure_port_clock_output(p_pll_clk, clk_pll_sync);
>    start_clock(clk_pll_sync);
>#endif
>2.
>#if !(defined(SPDIF_RX) || defined(ADAT_RX))
>/* Choose a frequency the xcore can easily generate internally */
>//#define PLL_SYNC_FREQ 1000000
>#define PLL_SYNC_FREQ 94000//1000000    //94000 for no driver,86000 no use
>#else
>#define PLL_SYNC_FREQ 300
>#endif

if so, I can sum up and whrite a few procedure/instruction for using this project.

Thanks again

doumdi, 2020-12-31:
@SCWhite. No, those changes should not be done in audiohw.xc and were done to test clock signals. I suggest you add information to a new markup file (ASSEMBLY_NOTES.md?) and make a pull request.

doumdi, 2020-12-31:
@SCWhite, I have also added the diff file in the master branch for main.xc in the usb_audio_module.

closed 2021-01-04 by SCWhite

jiangjiang4563, 2023-05-10:
hi, sir, did you get it to work properly and do you know if a computer supports two such devices at the same time

SCWhite, 2023-05-10:
@jiangjiang4563  Yes, I can confirm this project works properly (even magnificently)
and a normal computer (in my case linux and windows) are able to support more than one device at the same time.

jiangjiang4563, 2023-05-10:
> @jiangjiang4563 Yes, I can confirm this project works properly (even magnificently) and a normal computer (in my case linux and windows) are able to support more than one device at the same time.

jiangjiang4563, 2023-05-10:
@SCWhite Thank you for your prompt reply. It will keep me engaged in this work  :-)